HEP-SoC/SoCMake #21: Improvements to the TMRG function

mksoc, 2024-08-22:
The TMRG function should replace the sources keeping the correct "category", e.g. putting simulation only/FPGA only triplicated files in their respective language properties.

Also, it might be useful to be able to manually specify files to pass with the `-l` option, in order to keep the module definitions but strip the body.

mksoc, 2024-09-18:
Related to the second point, maybe we could use `vhier` to get only the used files, to avoid passing useless stuff to TMRG.

Risto97, 2024-09-18:
Maybe I am missing something, but if the files are not used, why are they listed?
Could you not use vhier to see which files are used and just remove them in CMakeLists.txt file?

Risto97, 2025-06-18:
This will be resolved with FILE_SETS #141 

If its not the case please reopen the issue @mksoc @nookfoo

closed 2025-06-18 by Risto97