HEP-SoC/SoCMake #39: ip_link

benoitdenkinger, 2024-09-12:
The ip_link function is not always able to find the dependency, it seems to scope/stack change and my dependency is not found. This is happening when I try to link an IP create by a dependency (through CPM package fetching) to a local IP (not created in my top cmake file). To solve this, I have to create a new IP locally (in my top cmake file) that is linked to the two IPs I want to link.

Side remark:
Many functions are using the flatten_graph one to get the dependencies (e.g., get_ip_include_directories, get_ip_sources), but this function is not commented and hard to understand. Why we do not use the INTERFACE_LINK_LIBRARIES property to get an IP dependencies as this is actually the property used by ip_link?

Risto97, 2024-09-12:
1.
This is weird, and it should not be happening.
I don't think its the scope issue, as the libraries added by `add_library()` calls have a `global` scope, so it doesn't apply the same rule as the variables.
It would be helpful if you could create a minimal example and I can fix it and add a test to regression.

2.
Regarding `flatten_graph()`.
I will have to take a look at it and possibly rewrite it to make it more readable.
I was hoping that the recent update in CMake with property inheritance on `INTERFACE_<PROPERTY>` properties would deprecate `flatten_graph()`, but I still think the current approach gives more flexibility, because the file lists are contained in each of the IP block libraries, so if you want to get list of files of only that particular level of IP block hierarchy, you can do it.
With property inheritance it will not be possible.

Property `INTERFACE_LINK_LIBRARIES` contains the dependencies of the IP block, but only 1 level below, not all the subdependencies. This way `INTERFACE_LINK_LIBRARIES` forms a tree structure of dependencies.

What `flatten_graph()` does is traverse this tree, and updates the `FLAT_GRAPH` property of the IP.
This property contains the whole dependency tree in a flat list that respects the hierarchy, making it easier to extract the file list with a simple `foreach()` loop.
This is important for compilers that expect submodules to be processed before modules that instantiate them (like PeakRDL, SV simulator for `packages`)

Risto97, 2024-10-04:
Hello @benoitdenkinger .

Indeed you were right, this seems to be an issue.
However, the issue is in CMake and not in SoCMake.

I opened an issue on their repository: https://gitlab.kitware.com/cmake/cmake/-/issues/26354
I think I can provide a workaround in the meantime until they fix it.

Risto97, 2024-10-04:
Hello @benoitdenkinger ,

Indeed that was a bug, It happens when the IP block is defined in one subdirectory, but `ip_link()` is called on that IP in different subdirectory.

I thought it was a bug, but it is intended operation of CMake (https://cmake.org/cmake/help/v3.30/prop_tgt/INTERFACE_LINK_LIBRARIES.html)

Pull request #84  explains a bit more

closed 2024-10-04 by Risto97