HEP-SoC/SoCMake #42: Consider shortened argument for add_ip

Risto97, 2024-09-12:
Right now `add_ip()` function has a following format:

```
add_ip(uart
    VENDOR company
    LIBRARY ip
    VERSION 0.0.1
    )
```

It is a bit verbose.
It would be possible to have a shortened version:

```
add_ip(company::ip::uart::0.0.1)
```

P.S.
There is already a possibility to omit VENDOR, LIBRARY, and VERSION.
However this would issue a warning.
```
add_ip(uart)
```

What are your thoughts @mksoc @benoitdenkinger @nookfoo ?

mksoc, 2024-09-12:
Probably not the first priority, but yes I like the shorter version!

Risto97, 2024-09-12:
I think I will make a `dev` or next release branch, and just get all the new feature and changes there, so I don't mess up your flow before you tapeout

commit 532d0781 (2024-09-14): allow add_ip short form, resolves #42 (#56)
closed 2024-09-14 by Risto97