HEP-SoC/SoCMake #44: Use generator expressions for conditional flags

Risto97, 2024-09-12:
Generator expressions can be used to simplify some of the backend functions.

For example if a function has following optional parameters:

```
    cmake_parse_arguments(ARG "XML;FILE_LIST" "TOP_MODULE" "" ${ARGN})
```

Instead of doing

```
    if(ARG_FILE_LIST)
        set(FILE_LIST_ARG --module-files)
    endif()

    set(__CMD ${VHIER_EXECUTABLE}
        ${FILE_LIST_ARG}
    )
```

We can do this:

```
    set(__CMD ${VHIER_EXECUTABLE}
        $<$<BOOL:${ARG_FILE_LIST}>:--module-files>
    )
```

These expressions can be more complex [generator expressions](https://cmake.org/cmake/help/latest/manual/cmake-generator-expressions.7.html#conditional-expressions)

mksoc, 2024-09-18:
I tried to use it in [`copy_rtl_files.cmake`](https://github.com/HEP-SoC/SoCMake/blob/master/cmake/utils/copy_rtl_files/copy_rtl_files.cmake) like so, but for some reason it doesn't work and returns always empty strings (I'm doing something wrong for sure).

```
    set(__CMD ${Python3_EXECUTABLE} ${CMAKE_CURRENT_FUNCTION_LIST_DIR}/copy_rtl_files.py
        $<$<BOOL:${ARG_TOP_MODULE}>:--top-module ${ARG_TOP_MODULE}>
        $<$<BOOL:${ARG_SKIPLIST_FILE}>:--skiplist ${ARG_SKIPLIST_FILE}>
        $<$<BOOL:${ARG_SYNTHESIS}>:--synthesis>
        --deps_dir ${FETCHCONTENT_BASE_DIR}
        ${INCDIR_ARG}
        $<IF:$<BOOL:${ARG_OUTDIR}>,--outdir ${ARG_OUTDIR},--outdir ${CMAKE_BINARY_DIR}/ip_sources>
        ${RTL_SOURCES}
    )
```

Risto97, 2024-10-05:
Hello @mksoc 

I was playing a bit more with the generator expressions, the issue seems to be having whitespaces in the generator expressions.
It is explained here: https://cmake.org/cmake/help/latest/manual/cmake-generator-expressions.7.html#whitespace-and-quoting

To summarise for your example, you need to:

* Wrap the generator expression with quotes ""
* Instead of using space, use escaped list symbol \\;
* Add **COMMAND_EXPAND_LIST** argument to the add_custom_command() call

Your command would look like this:

```
    set(__CMD ${Python3_EXECUTABLE} ${CMAKE_CURRENT_FUNCTION_LIST_DIR}/copy_rtl_files.py
        "$<$<BOOL:${ARG_TOP_MODULE}>:--top-module\;${ARG_TOP_MODULE}>"
        "$<$<BOOL:${ARG_SKIPLIST_FILE}>:--skiplist\;${ARG_SKIPLIST_FILE}>"
        "$<$<BOOL:${ARG_SYNTHESIS}>:--synthesis>"
        --deps_dir ${FETCHCONTENT_BASE_DIR}
        ${INCDIR_ARG}
        "$<IF:$<BOOL:${ARG_OUTDIR}>,--outdir\;${ARG_OUTDIR},--outdir\;${CMAKE_BINARY_DIR}/ip_sources>"
        ${RTL_SOURCES}
    )
```

Also keep in mind that the the generator expressions are evaluated at the generate phase, after the configure phase.
This means that if you use it for **OUTDIR** variable, and later if you build a file path using that variable (file generated by a command), in the <LANGUAGE>_SOURCES of the IP there will be a generator expression at configure time, which might be a bit annoying.
I would maybe not use generator expression for OUTDIR or anything that influences file paths, at this point.

You can decide if you think its worth using them for simple CLI options, I am not sure at this point its better than having an if() clause.

Risto97, 2025-06-18:
Generator expressions should be used where it makes sense, no need to refactor the codebase for this.
So closing the issue

closed 2025-06-18 by Risto97