HEP-SoC/SoCMake #116: Detecting source-header dependencies for Verilog

Risto97, 2025-01-09:
Currently, there is no way to retrigger targets if the verilog header files change.
This is a similar problem to how C++ header files are tracked with depend.make in Makefiles.
 
Currently, easiest solution to implement is to list the header files manually, something like:

```
ip_sources(${IP} SYSTEMVERILOG
    HEADERS 
     file1.sv
....
```

Risto97, 2025-02-09:
This is implemented in 031850799ed383f3a080eb6f59e1c9849728ee50.

I would appreciate feedback on this feature, or maybe alternative ideas on how to solve it @mksoc @benoitdenkinger @nookfoo 

To add more context.
In SystemVerilog-UVM components are implemented as SV classes.
Classes are usually written in Header files, not in source SystemVerilog files.

Simulators expect only source files to be passed as CLI arguments, while header files are expected to be found through include directories, passed as (+incdir+<dir-with-header>).
Since in SoCMake so far we only have the include directories of header files set with `ip_include_directories(${IP} SYSTEMVERILOG ...)` there is no way to track the changes of said header files and retrigger builds.

The solution found in CMake and C++ (gcc depend files) is not available in SystemVerilog tools, so at the moment I opted for the option of explicitly listing header files.

There are some advantages to this solution:
- Having them explicitly listed would allow us to do more easily tasks such as installing, linting, or any other custom step as the file list is available at configure time.
- Simplicity, not having to rely on a commercial or a custom tool to extract dependencies will make it less error-prone.

A disadvantage is:
- Once the header file changes a whole IP block would be recompiled, even IP blocks that depend on it, as opposed to C++ where only the depending source file gets recompiled. This might not be a big issue, as it is anyways difficult to do per-source file compilation with available simulators.

With this implementation header files can be added as follows:
```
ip_sources(${IP} SYSTEMVERILOG
      source1.sv
      source2.sv
    HEADERS
      header1.sv
      header2.sv
    )
```

`get_ip_sources(sources ${IP} SYSTEMVERILOG}`
Would still return only `source1.sv` and `source2.sv`

To get the header files only (without sources) call:
`get_ip_sources(sources ${IP} SYSTEMVERILOG HEADERS)`

P.S. There was an alternative implementation of this, to be more consistent with CMake.

CMake you can list header files (optional) in the target_sources() function like this:
```
target_sources(test PUBLIC
   main.cpp
   hello.cc
   hello.h
   hello.hh
  )
```
There are no delimiter `HEADERS`, but instead CMake detects the file extensions.
I decided not to go this way because:
1. This requires a predefined set of extensions for languages, right now SoCMake is pretty flexible on adding new languages, and this would require more work when adding a new one.
2. What if users use some weird extension for SystemVerilog?

benoitdenkinger, 2025-02-10:
Tracking header files is definitely helpful. We faced the problem when "exporting" the sources from a target but only having the header files include directories. What we implemented was copying all the files inside the include directories, which is not ideal.

Regarding the implementation, I don't have a strong opinion. Supporting the non-explicit solution (without the HEADERS parameter) requires defining standard extensions, but I guess it is the same for the target_sources() built-in function. At some point, we cannot support any input from users and I've never seen SystemVerilog files with an extension different that 'sv' or 'svh' (but maybe for other languages it is not so standard). I think this also imposes some minimum rules, I wouldn't like (and provide support) to a project that defines SystemVerilog files with meaningless extensions.

benoitdenkinger, 2025-02-11:
### Simulator/tool specific files

Files like the vlt ones are simulator specific and each simulator have its own set of input files it can defined. For verilator, this `.vlt` files are configuration files. The extension `.vlt` is the default proposed one but what makes it a verilator configuration file is only the <code>`verilator_config</code> directive in the file (but they are not SystemVerilog files).

For this kind of files (simulator/tool specific), we either need to support them with an additional language (which is already what we have for these vlt files I believe) or we need to give the possibility to the user to add the extra language (that we have) and also put an extra generic argument when calling the different simulators (and tools in general) to be able to pass extra commands/files. If the files are commonly used (e.g., these vlt files), I would support them natively, for more custom features I would let the users do the work.

### Header installation

When calling the target_source() function, one format is ([source](https://stackoverflow.com/questions/78558637/how-are-file-sets-supposed-to-be-used-in-cmake)):

```
target_sources(
    file_loader
    PRIVATE
        src/file_loader.cpp

    PUBLIC
    FILE_SET HEADERS
    BASE_DIRS src/include
    FILES
        src/include/lib_file_loader/file_loader.h
)
```
In `src/file_loader.cpp` you do `#include "lib_file_loader/file_loader.h"`

We could also request the BASE_DIRS with the HEADERS (I'm not proposing to use the other syntax format with the FILE_SET, this is just an example I found).