HEP-SoC/SoCMake #123: Separate file set for simulation, synthesis, etc...

Risto97, 2025-02-12:
As was mentioned some of the build systems like tmake separate sources into different categories.

From what I understand these are usually:
 
* Simulation only (these are non-synthesizable files)
* Synthesis only (probably macro cells)
* RTL files (standard RTL files that work for both)
* Anything else (TMRG)???

First, it is important to define well what are these file sets, can we even predict all of them?

This could be implemented in different ways:

--------------------------------

### 1. Define a new language for the file set with the current `ip_sources()` implementation
```
ip_sources(${IP} SYSTEMVERILOG
   [RTL files ...]
   )
ip_sources(${IP} SYSTEMVERILOG_SIMULATION
   [SIM files ...]
   )
ip_sources(${IP} SYSTEMVERILOG_SYNTHESIS
   [SYNTH files ...]
   )
```

Later to retrieve the list of sources, we need to do the following:
```
get_ip_sources(SOURCES ${IP} SYSTEMVERILOG SYSTEMVERILOG_SIMULATION SYSTEMVERILOG_SYNTHESIS)
```

Pros:
* It allows for all sources to be present within the project configuration
* Easy mapping to build systems such as tmake
Cons:
* I am not confident we can predict all necessary categories, does SystemRDL also have these categories?
* For all the new languages, we need to add them in check_language() function
* All EDA tools that use these files will be affected, and the `get_ip_sources()` call will need to be changed in each of them
* (Subjective) In my opinion this separation is a bit confusing

--------------------------------

### 2. Use FILE_SETS that will be implemented in the future for the purpose.
The syntax would be a bit cleaner, would look something like this I assume:

```
ip_sources(${IP} SYSTEMVERILOG
   [RTL files ...]
   FILE_SETS
        SIMULATION    [SIM files ...]
        SYNTHESIS     [SYNTH files ...]
        ....
   )
```
Pros stay the same as previous
Cons are mostly the same.

--------------------------------

### 3. Use configuration options

This feature is already present, and it's my prefered way of handling this, as it is the easiest to understand

```
option_boolean(SYNTHESIS "Synthesize design" OFF)

ip_source(${IP} VERILOG
    [RTL files ...])

if(SYNTHESIS)
   ip_sources(${IP} VERILOG
      [SYNTH files ...]
   )
else()
   ip_sources(${IP} VERILOG
      [SIM files ...]
   )
endif()
```

Pros:
* It reuses a feature already present in SoCMake to achieve the same result, keeping the number of features in SoCMake lower (easier to maintain)
* All the flexibility of defining categories is left to the user, I prefer not to impose workflows, but keep it flexible
* EDA backend functions are agnostic of these categories, and no changes are necessary
Cons:
* Not all the files are available on every project to configure, which can be a problem for the installation (we can argue that CMake also separates Release and Debug builds similarly, you can't have both)
* More difficult to export to other build systems, such as tmake

--------------------------------

### 4. add_ip() can have special IP categories

Instead of separating these files per FILE_SETS, separate IP blocks per categories:

```
add_ip(i2c RTL)
ip_sources(i2c 
   [RTL sources ...]
   )

add_ip(i2c_synth SYNTHESIS)
ip_sources(i2c_synth 
   [SYNTH sources ...]
   )

add_ip(i2c_sim SIMULATION)
ip_sources(i2c_sim 
   [SIM sources ...]
   )

ip_link(i2c
   i2c_synth
   i2c_sim
)
```

This has similar PROS and CONS as 1,2. I think its a different flavor of the same solution.

--------------------------------

### 5. ip_link() can have different link types.

In CMake there are already types of linkage as INTERFACE, PUBLIC, PRIVATE, that affect the visibility of properties.
Similar could be done for ip_link() for IP categories

```
add_ip(i2c)
ip_sources(i2c 
   [RTL sources ...]
   )

add_ip(i2c_synth)
ip_sources(i2c_synth 
   [SYNTH sources ...]
   )
ip_link(i2c SYNTHESIS i2c_synth)

add_ip(i2c_sim)
ip_sources(i2c_sim 
   [SIM sources ...]
   )
ip_link(i2c SIMULATION i2c_sim)
```
This has similar PROS and CONS as 1,2. I think its a different flavor of the same solution.

Risto97, 2025-02-12:
Let me know  if you have any other idea, or want to add PROS or CONS for some solution

benoitdenkinger, 2025-03-11:
Hi @Risto97, nice summary of the various options. As mentioned, the main difference is to embed these properties inside SoCMake (all options except 3) or not (option 3).

I like option 3 because it leaves the choice to the user. Although you mention as a drawback the limited compatibility with other build system like tmake, I don't think we should care too much about that. The user can always add its own properties. For example:

```
option_boolean(SYNTHESIS "Synthesize design" OFF)

ip_source(${IP} VERILOG
    [RTL files ...])

if(SYNTHESIS)
   ip_sources(${IP} VERILOG
      [SYNTH files ...]
   )
else()
   ip_sources(${IP} VERILOG
      [SIM files ...]
   )
endif()

# tmake integration
ip_sources(${IP} TMAKE_VERILOG_SYN)
ip_sources(${IP} TMAKE_VERILOG_ENV)
ip_sources(${IP} TMAKE_SYSTEMVERILOG_TB)
...
```
In this case, only the user custom functions, which generate files for tmake, will check for the properties TMAKE_<...>. It duplicates a bit the cmake code, but I think it is fine.

Option 3 is also the less restrictive and it will be easy to keep backward compatibility if another approach is taken at some point. I'll try to play with it with our project and see if I find any problems.

benoitdenkinger, 2025-03-11:
Still, option 3 will impose a certain standard, especially for open-source project, as people will need to define the same flow variables (e.g., SYNTEHSIS, FPGA) to be able to share or reuse existing IPs.

Risto97, 2025-03-11:
Hello @benoitdenkinger ,

> Option 3 is also the less restrictive and it will be easy to keep backward compatibility if another approach is taken at some point. I'll try to play with it with our project and see if I find any problems.

Exactly, I think this is the most important. We don't need to take this decision now and come up with something suboptimal.
The current solution with configuration options works well for what we need.
I would rather add this feature later than have to remove a bad implementation.

I think the biggest issue of option 3 is IP block installation, and once we start working on that, it will clarify a bit more what kind of approach we should take.

benoitdenkinger, 2025-04-17:
After thinking about this, I have come up with another proposal which is very close to option 1.

### Option 1b

By default, each HDL language (Verilog, SystemVerilog, VHDL) is listed under its own <LANGUAGE>_SOURCE property and the various simulators (or other functions) always fetch these source files. In addition, every function fetching target sources can have a FILE_SETS argument that lists additional languages to fetch.

For example, the user can define any file under one of the three basic type and the files listed under them will always be fetched when getting the sources of an IP:

```
ip_sources(myIP [SYSTEMVERILOG,VERILOG,VHDL]
   [RTL files ...]
)
```

Then, the user can also define additional type of sources (with any name):

```
ip_sources(myIP SYSTEMVERILOG_MODEL
   [RTL files ...]
)

ip_sources(myIP SYSTEMVERILOG_SIM
   [RTL files ...]
)

ip_sources(myIP SYSTEMVERILOG_SYNTHESIS
   [RTL files ...]
)

ip_sources(myIP SYSTEMVERILOG_FPGA
   [RTL files ...]
)
```

When running RTL simulation (or any other function) you can do:

```
xcelium(myIP) => this fetches only the files listed in SYSTEMVERILOG, VERILOG, and VHDL
```

or you can add any fileset:

```
xcelium(myIP
    FILE_SET SYSTEMVERILOG_SYNTHESIS SYSTEMVERILOG_FPGA
)
```

This solution

Risto97, 2025-04-17:
Hello @benoitdenkinger ,

I have one issue with this, for example when you call XCelium like this

```
xcelium(myIP
    FILE_SET SYSTEMVERILOG_SYNTHESIS SYSTEMVERILOG_FPGA
)
```
The xcelium function also compiles VHDL, but should it ignore VHDL files in this case?

I think the option 2 is very similar, but its more clear about this issue, for example:

```
ip_sources(myIP [SYSTEMVERILOG,VERILOG,VHDL]
   [RTL files ...]
)

ip_sources(myIP SYSTEMVERILOG
   FILE_SET MODEL
       [RTL files ...]
   FILE_SET SIM
       [RTL files ...]
)

ip_sources(myIP VHDL
   FILE_SET MODEL
       [RTL files ...]
   FILE_SET SIM
       [RTL files ...]
)
```

When running xcelium for a particular file set
```
xcelium(myIP
    FILE_SETS MODEL SIM
)
```
Now it's clear that the MODEL and SIM file sets should be used also for VHDL, VERILOG and SYSTEMVERILOG targets.
It also feels a bit easier to map to IPXact fileSet elements.

What do you think about this one?

benoitdenkinger, 2025-04-28:
Yes, indeed, your proposal is more clear.

Initially I was more into option 3 (using configuration options) and I still think this is the safest solution. However, option 3 is only a temporary solution. I find your last proposal / option 2 more clear and elegant. At the end, we have to make this distinction between these file sets so I think it is better to support it natively and option 2 still give enough flexibility.