HEP-SoC/SoCMake #126: Mixed language single-kernel SC/SV/VHDL

amal-khailtash, 2025-02-15:
Are there any flows where I can mix SystemC with SystemVerilog/VHDL in a single kernel (vendor compile)?  Use cases might be SystemC (UVM-SC) on top or SystemVerilog (UVM) on top.

The current mixed language example assumes SV/VHDL only.

amal-khailtash, 2025-02-16:
Basically, what I am looking for is how to do the equivalent [simple_sc_verilog](https://github.com/Risto97/VeriSC/tree/master/examples/simple_sc_verilog) and also simple_sv_sc version using SoCMake.

Risto97, 2025-02-16:
Hello @amal-khailtash ,

Thank you for taking an interest in SoCMake.

Yes, this is possible, and there was already a more complete example here: https://github.com/HEP-SoC/apb_crc_uvm_socmake/tree/master/sc-uvm
It includes a full SystemC-UVM testbench (missing CRAVE for constrained random, and functional coverage, which is also possible)

I ported quickly the example you suggested from VeriSC here (https://github.com/HEP-SoC/SoCMake/tree/develop/examples/simple_sc_sv).

Note that there are some limitations currently, it is only possible to have SystemC on top, SV/Verilog as DUT and only with Verilator.

I have in the roadmap to implement also what you suggested, that is:

1. SystemC on top with commercial simulators (only Verilator supported now)
2. SystemVerilog on top, SC as dut with commercial simulators (For Verilator this would not be possible)
3. SystemC on top, VHDL as dut, for commercial simulators
4. Possible mix of hierarchy levels SC-SV-SC... SV-SC-SV with commercial simulators

Which simulator are you interested in using for SV, Verilog, and VHDL? Would Verilator work for now?

To give you a short history of the project.
SoCMake was started for this exact purpose, doing SystemC verification. VeriSC was the initial attempt at the topic and was since archived in favour of SoCMake.

After a while, I realized that SoCMake could do much more than SystemC verification, so I started developing it as a generic HW/SW build system.
Unfortunately, I started prioritizing other aspects of the build system, and the SystemC verification features from VeriSC were not fully ported, hence no support for SC-SV simulation with commercial simulators.

I do plan to bring these features back into SoCMake in the next months, however, I am limited with time, and I am trying to polish core aspects (IP library API) for the 1.0.0 release.

If you are interested in helping with the development, it would be very appreciated.
It would also be helpful if you could share your use case.

amal-khailtash, 2025-02-17:
Hello @Risto97, thanks for pointing to that depo as well as porting the example.

Unfortunately, my use case does not allow Verilator.  We mainly use Synopsys VCS and sometime Questasim as well, rarely Xcelium.

We have a mix of testbenches.  Mostly UVM-SV or UVM-SC on top.  As we use HLS for some designs, our DUT can be either SC or SV.  The DUT SC would have an equivalent SV (with possibly some port adaptation).  As we are mainly using VCS, we are thinking of moving away from SC TB, because of a limitation with VCS that they call Donuts.  VCS does not support crossing SC->SV->SC.   And this prevents us from replacing our lower-level SC designs with SV.  So, we might need to go to SV (TB) -> SV (DUT) -> SC -> SV.  We do not need to worry about this now, but the idea would be to use mixed-language single-kernel compile under VCS and allow mix and match top/bottom blocks if possible.  We do not use VHDL much, unless there is a vendor IP we need to use.

It is great that you separated the build flow from actual HLV concerns.

We mostly use Makefiles and our custom build scripts, and I was looking to implement this with CMake as I find it is modularized,  extendable.  We also have a mix of C/C++ and DPI code that also needs to be compiled and I thought CMake is a good choice.

Thinking there must be some extension out there, I came across this repo.

I can certainly help time permitting.  My CMake knowledge is rusty and not extensive.

Risto97, 2025-02-17:
Hello @amal-khailtash ,

Thank you for sharing this information.

Yes, I remember some limitations with VCS.
However the support will come gradually, I will first support SV-SC and SC-SV, the other combination might or might not come for free after that.

There is one thing that I am trying to do that makes it a bit more difficult, and that is avoiding to use of vendor tools that wrap GCC calls to compile and link C++ executables.
Instead of using tools such as syscan, xmsc, etc..., I am trying as much as possible to understand how to use GCC/Clang toolchains directly.
This allows for a bit more reusability between the support for different tools, and better CMake integration, but also makes it more difficult, as sometimes some reverse engineering is needed.

What you are trying to do, sounds very similar to what I wanted when I started the project, and I was in a similar situation picking which language/build system to base it on.
In the end, I think CMake is a good platform for this solution and I was also surprised there was no similar project implementing this.
From my experience, Makefiles do not scale well for bigger projects, as there is a lack of reusability, and would have to reimplement a lot of features already present in CMake.

For DPI-C, this is a feature SoCMake supports, you can take a look at how it is loaded into the SV-UVM testbench.
First, the DPI library is compiled as a shared (or static) library here: https://github.com/Risto97/apb_crc_uvm_socmake/blob/master/sv-uvm/crc_dpi/CMakeLists.txt, note also how easy it is to fetch an external C++ library with CPM and link it in a CMake project.
Later in SoCMake, linking the DPI library into the SV testbench can be done using `ip_link()` function: https://github.com/Risto97/apb_crc_uvm_socmake/blob/master/sv-uvm/CMakeLists.txt#L36.
There is a similar example also for VHDL UVVM here: https://github.com/HEP-SoC/UVVM_SoCMake/blob/master/examples/apb_crc/CMakeLists.txt

I hope to get work restarted on this in the next weeks, and if you are interested you can give it a try and give me feedback. This would already be a great contribution and help for me.

amal-khailtash, 2025-02-17:
To be able to used mixed language simulation, in VCS, one needs to use `syscan` at least on a dummy file.  Whether one compiles SC files with g++ outside of VCS or not.  Using vendor compile commands (VCS: syscan), (Questa: sccom), ... allows them to create wrappers and translations between SC and SV port types.

There are strict dependence on specific g++ versions for both VCS and questa if one wants to use mixed, single-kernel simulation.

I can certainly help with reviews or implementaion.

Risto97, 2025-02-23:
Hello @amal-khailtash ,

I had a little bit of time to work on this.
Yes, you are right, the vendor tools cannot be fully avoided.

I was playing a bit with Questa: sccom, scgenmod commands. Although I could manage to compile the object files with G++, and to link them, by looking at how they invoke g++ when passed `-vv` flag.
These commands do more than just invoke g++, and it's mostly updating the simulation database, for `vsim` to later find compiled modules.
I had some success by passing `-nocc` flag during `sccom` compilation, where only the database is updated but object files are not compiled. However this flag does not work for `-link`, when the object files are linked into `systemc.so` library.

To start the conversation, I made a "naive" implementation of SystemC-Verilog mixed language simulation.
I execute `sccom`, and do not try to compile object files with g++, for now.
The plan is to get all the simulators working with a simple example first, utilizing vendor tools, and then later try to extract similarities between them, and hopefully replace SystemC compilation with g++.

Here is an example that I prepared for testing what is currently supported:
https://github.com/HEP-SoC/SoCMake/blob/b7531fb0ab14f949111ca0b080f4e9ef898daa97/examples/simple_mixed_language_sc_vlog/CMakeLists.txt#L85-L138

This example currently supports following scenarios:

* **OSCI_SC**: Open-Source SystemC only tesbench, compiled with GCC
* **VERILATOR_SC_VLOG**: OSCI SystemC on top, and verilated Verilog as submodule 
* **MGC_SC**: Pure SystemC using Questa's shipped SystemC shared library
* **MGC_SC_VLOG**: SystemC on top, verilog as submodule using Questa `sccom` and `scgenmod`
* **MGC_VLOG_SC**: Verilog on top and SystemC as submodule, using Questa
* **MGC_VLOG**: Pure verilog using Questa

To run the simulation, execute the following steps:

```
git clone -b modelsim_sc_sv https://github.com/HEP-SoC/SoCMake.git
cd SoCMake/examples/simple_mixed_language_sc_vlog
mkdir build && cd build
cmake -DSCENARIO=<SCENARIO-NAME> ../
make help_all
make <run-target-name>
```

---------------------------------------------------------------

I would be interested in hearing your feedback on this flow. 
I see that you have an idea of using a dummy file to trick the simulator to generate databases to use g++ independently.
Would you be interested in discussing this over a call, it would be easier for me to understand your use case, and hopefully we can steer the implementation in the right direction.

amal-khailtash, 2025-02-24:
Hi @Risto97, I did try a few of your scenarios.  Some passed and some failed.  I can give you more feedback once I try them all and resolve my compile/link issues with gcc/libraries incompatibilities.

You can add to your network on LinkedIn (on my github profile) and we can swap contacts.  I can definitely give you more feedback.

amal-khailtash, 2025-03-03:
Here are a few general recommendations:

1. In the mixed language example, it would be good if you could separate the design and test into their own directories.  Also, have separate directories for different implementation (SC vs SV).
2. I am not familiar with the decisions that made into the organization of CMake targets.  Normally we try to separate the DUT compile from TB compile for one: different teams handle those and two: faster more efficient compile (compile DUT once and TB as it gets developed).
3. I would separate SCENARIO into SIMULATOR (OSCI, MGC, VERILATOR, VCS, ...) and then topology differently.  I will let users be able to pick and choose the underlying implementation: SC vs SV, ...  Logistics need to be hashed out.
4. Separating by IP in the build directory might be useful, but also allowing users to compile into different libraries would be great too.
5. Other that Topology (SC on top or SV on top), I would use of SystemVerilog configuration to select the implementation of the different lower level blocks and switch SC vs SV based on SV configuration.  Or otherwise allow the selection in CMAKE.

amal-khailtash, 2025-03-03:
6. Provide a way to run simulation using interactive GUI.

When I build for `OSCI_SC`, I do not see any build targets for compile/sim:
```
Target                      Description
----------------------------------------------------------
graphviz                    Generate a build graph with graphviz
help_ips                    Print IPs help
help_targets                Print targets help
help                        CMake native help
----------------------------------------------------------
```
So, I just ran `make`, but SOCMake assumes SystemC library are available here!

`apb_crc_uvm_socmake/sc-uvm/build/systemc-build/src/systemc-build/src/libsystemc.so.2.3.4`

7. Would be great if I could specify SYSTEMC_HOME for include/library.  I am not sure how to override the above path.
8. I have also seen some usage of [CONAN2](https://docs.conan.io/2/) for dependency management.  Maybe we can use it for SystemC and other dependencies as well.

amal-khailtash, 2025-03-03:
10. It would be great if we allow users to bring in SOCMake CMakefiles in a way that does not make assumptions on directory structure and let users bring in SOCMake easily into their project.

Risto97, 2025-03-03:
Hello Amal,

Thank you, these are some excellent points!
I will try to explain one by one.

### 1. Split the IPs into subdirs

I have split the design and test into separate directories.
Further, I have split the DUT into separate IPs `counter1`, `counter2` and `counters` IP libraries, to demonstrate some later points you made.
Their definition resides in a separate directory.

```
simple_mixed_language_sc_vlog/
├── CMakeLists.txt          # Top level CMake file
├── counter1
│   ├── CMakeLists.txt      # counter1 IP definition
│   ├── sc
│   │   └── counter1.h
│   └── vlog
│       └── counter1.v
├── counter2
│   ├── CMakeLists.txt      # counter2 IP definition
│   ├── sc
│   │   └── counter2.h
│   └── vlog
│       └── counter2.v
├── counters
│   ├── CMakeLists.txt      # counters IP definition
│   ├── sc
│   │   ├── counters.cpp
│   │   └── counters.h
│   └── vlog
│       └── counters.v
├── deps
└── tests
    ├── CMakeLists.txt      # test_counters IP definition
    ├── sc
    │   ├── test_counters.cpp
    │   └── test_counters.h
    ├── sc_main.cpp
    └── vlog
        └── test_counters.v
```

This way in the top level `CMakeLists.txt` we can define only the targets (simulation, etc...).
While the IP level `CMakeLists.txt` is simulator agnostic, and only contains the definition of IP.

The hierarchy (in the case of modelsim simulation) looks like this:

[image]

----------------------------

### 2. CMake compile targets

Currently, when using commercial simulators the compilation is split, but not per CMake `custom_target`, but rather per `custom_command`, when you run `make run_test_counters_modelsim` in this example, the following commands are executed:

```
$ make run_test_counters_modelsim
[ 20%] Compile SV, and Verilog sources of counter1 with modelsim vlog in library work
[ 40%] Compile SV, and Verilog sources of counter2 with modelsim vlog in library work
[ 60%] Compile SV, and Verilog sources of counters with modelsim vlog in library work
[ 80%] Compile SV, and Verilog sources of test_counters with modelsim vlog in library work
```

This follows the IP dependency graph, where each IP will be compiled in parallel, as a separate command.
But as you can see on the above graph, only 1 compilation target exists `test_counters_modelsim_complib`.

I used to have a function argument for creating a separate target for compiling each of the IP blocks, but I decided to remove it as I thought it would create too many targets for big designs.
It was used something like this:
[... 266 lines cut ...]
cmake -DSIMULATOR=mgc -DTEST_LANG=systemc -DDUT_LANG=systemc ../
make run_test_counters_modelsim
```

I see the same issue as before so hopefully `LD_LIBRARY_PATH` fixes it.
However, I see that there might be also an issue with 32bit vs 64bit simulations, I commented out the -64bit simulation argument on this branch.
As you are using RH8 and I am using RH7 there is a mismatch in the default bitness of the systems, so we might get different results.
I will need to understand this problem and try to solve it in a generic way.

------------------

### VCS support

I was taking a break for a few days, but I did get in the meantime support for XCelium.
With XCelium I had much better luck with avoiding the vendor-specific commands, and I can compile and link the design with GCC.
I will push it once I finalize the support and adapt it to the example with counters.

After that, I plan to add also preliminary VCS support.
Then if you have time, you can review the flow, I will contact you on Linkedin to potentially schedule a call, as it might be easier than typing long messages like this one.

-----------------

Thank you very much for the excellent feedback and advice.
This contribution is very important for the project.
I added you to the contributor list, let me know if you are ok with this or if you need to add attribution for your company.

amal-khailtash, 2025-03-04:
This is great!  I hit a few issues and I have some more comments.

1. For OSCI_SC:

When I specified `-DSYSTEMC_HOME=<path-to-sc>` it could not find SystemC and started downloading, building.  Not sure why.
```
-- Could NOT find SystemCLanguage (missing: SystemCLanguage_DIR)
-- [SystemC Not Found]
-- [Building SystemC]
```

Also, I was expecting the SystemC `deps` be under the build directory, but they seem to be in the example root directory.

For those who want to download and build SystemC, they should also be able to specify SystemC version to download build instead of the default 2.3.4.

I am not sure if you are using VSCode or not.  VSCode has CMake integration and it can show you the cmake targets.  I was hoping I see the cmake targets, but this is what I see:

[image]

This is a nice to have if we could see the configured targets.

2. For VERILATOR_SC_VLOG I see:

```
CMake Warning (dev) at CMakeLists.txt:4 (find_package):
  Policy CMP0144 is not set: find_package uses upper-case <PACKAGENAME>_ROOT
  variables.  Run "cmake --help-policy CMP0144" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  CMake variable VERILATOR_ROOT is set to:

    <PATH-TO-VERILATOR>/share/verilator/include/../

  For compatibility, find_package is ignoring the variable, but code in a
  .cmake module might still use it.
This warning is for project developers.  Use -Wno-dev to suppress it.
```

And I also see: `error: #error "Verilator requires a C++14 or newer compiler"` and a whole lot of C++ compile errors.  Maybe we are somehow missing -std=c++14?  Or is this specific to Verilator version I am using?

3. For MGC_VLOG:  Some warnings, but works otherwise.

```
# ** Note: (vsim-3812) Design is being optimized...
# ** Warning: (vopt-10016) Option '-L work' was detected by vlog for design-unit 'test_counters', but was not detected by vopt. The vlog option will be ignored.
# ** Warning: (vopt-10016) Option '-L work' was detected by vlog for design-unit 'counters', but was not detected by vopt. The vlog option will be ignored.
# ** Warning: (vopt-10016) Option '-L work' was detected by vlog for design-unit 'counter1', but was not detected by vopt. The vlog option will be ignored.
# ** Warning: (vopt-10016) Option '-L work' was detected by vlog for design-unit 'counter2', but was not detected by vopt. The vlog option will be ignored.
```

GUI mode works fine too, but you might want to consider supporting Visualizer.  Compile needs some more options.  I was thinking how to allow user to specify some custom compile/run options.  Should that all go to CMakeList.txt?

4. For MGC_SC_VLOG:

I do not usually like to pollute LD_LIBRARY_PATH and usually avoid it, or the executable needs to always know where the .so shared library is with LD_LIBRARY_PATH.

You might consider adding rpath using: `-L<QUESTA_HOME>/gcc-<GCC_VERSION>-linux/lib -Wl,rpath,<QUESTA_HOME>/gcc-<GCC_VERSION>-linux/lib -lsystemc`

This way, the path to the link libraries are placed in the executable.  I did not specify the LD_LIBRARY_PATH but I am getting a different error during link:

```
[ 66%] Built target test_counters_modelsim_complib
[ 83%] Link SystemC objects into systemc.so for test_counters with sccom
sccom -link -nologo 
Start time: 20:26:27 on Mar 03,2025
** Warning: (sccom-6197) No object files found for SystemC version 2.3.4 compiled with gcc-10.3.0 in 'work'. Skipping this library for "sccom -link".
** Error: (sccom-6207) No object files found for SystemC version 2.3.4 with gcc-10.3.0 in any library. Make sure that compilation and linking is being done with correct options.
End time: 20:26:28 on Mar 03,2025, Elapsed time: 0:00:01
```

I think the problem this time is that SystemC dependency is compile with a different gcc version 10.2.0 and when Questa sccom compiles, it uses their gcc version 10.3.0.  The SystemC library that is bundled with Questa is static version.

5. For MGC_VLOG_SC:
6. For MGC_SC:

I see the same issue as above.  Link issue and I don't think is related to -32 or -64, but something related to gcc version.

Risto97, 2025-03-05:
Hello Amal,

###  For rebuilding SystemC, it can happen for multiple reasons:

*  It could not find `SystemCLanguageConfig.cmake` file in your SystemC installation, its typically in this directory: `systemc-2.3.3/lib/cmake/SystemCLanguage/`
* The version might not be correct, as in `CMakeLists.txt` this is specified `systemc_build(VERSION 2.3.4 EXACT_VERSION)`,  it looks for the exact version `2.3.4` if it finds a different version it would reject it because of `EXACT_VERSION` argument. You can modify this line to match the version you want to use. There is no real reason I picked `2.3.4` for this example. Something that can help you debug `find_package()` CMake command is setting `cmake -DCMAKE_FIND_DEBUG_MODE=on ../` during configuration, CMake will print its procedure for finding packages. 

> Also, I was expecting the SystemC deps be under the build directory, but they seem to be in the example root directory.

This is set by `FETCHCONTENT_BASE_DIR` at the beginning of the `CMakeLists.txt`, you can read more about it [here](https://cmake.org/cmake/help/latest/module/FetchContent.html#variable:FETCHCONTENT_BASE_DIR).
By default the package manager would clone the repositories in the build directory, but with this variable you can override it to be cloned somewhere else.
The reason I use it is that I want to reuse the same dependencies sometimes between multiple `CMake` subdirectories.
Lets say you have 2 testbenches that are in 2 directories and have their own `CMakeLists.txt`, but they maybe both use `SystemC` and same IPs. With `FETCHCONTENT_BASE_DIR` you can assure that they clone and build it only once.

Another reason is that I might remove the build directory from time to time, this way the dependencies in `deps/_deps` will stay, so I would not have to recompile SystemC.

If you want it in the build directory, you can remove this line.

> For those who want to download and build SystemC, they should also be able to specify SystemC version to download build instead of the default 2.3.4.

Yes, I agree, however as you noticed not everything is available as CMake configuration option, you could very easily do something like this:

```
option_string(SYSTEMC_VERSION "Version of SystemC to use" "2.3.4")

systemc_build(VERSION ${SYSTEMC_VERSION} EXACT_VERSION)
```
SoCMake at this point does not try to impose too many opinions on a user flow, the `CMakeLists.txt` files are supposed to be written per project, and the user gets all the freedom to decide how their build flow works.
It is a deliberate that there are no functions like `simulate()` at this moment, instead there are functions for simulators like `modelsim(), xcelium(), vcs(), ...`. I consider option as `simulate()` to be a layer above the EDA tool layer we can call "flow" layer, and at this point, I am focusing on the EDA tool layer, rather than the "flow" layer.
Although a user can easily write a "flow" layer to their liking which might assume some configuration options by default.

Regarding VSCode, I do not use it, but it seems like a nice feature that users might want to use.
I will try to install it and see if it appears to me, does it appear for you on other CMake-based projects, but not for SoCMake ones?

###   For VERILATOR_SC_VLOG I see:

For the warning you see during compilation.
```
CMake Warning (dev) at CMakeLists.txt:4 (find_package):
```

Yes, I am aware of it, it's unfortunate that I am using `VERILATOR_ROOT` variable, which has a special meaning for `find_package()` CMake function, so it complains about it, but it should not affect anything.
I will fix this once I  do a refactor of `verilator` support soon. 

> And I also see: error: #error "Verilator requires a C++14 or newer compiler" and a whole lot of C++ compile errors. Maybe we are somehow missing -std=c++14? Or is this specific to Verilator version I am using?

This is weird, but I suspect it has to do with Verilator version and version of GCC you are running now. Can you tell me which Verilator version you are using with `verilator --version` and also `gcc --version`.
You can try to use the GCC provided with Questasim installation, to do that you can use `CMAKE_CXX_COMPILER` option when configuring CMake:
```
cmake -DSIMULATOR=osci -DTEST_LANG=systemc -DDUT_LANG=verilog -DSYSTEMC_HOME=<path-to-sc> -DCMAKE_CXX_COMPILER=<path-to-g++> ../
make -j16
./systemc_example
```
Make sure to remove the build directory before changing the compiler.

### For MGC_VLOG: Some warnings, but works otherwise.

I don't see the warnings that you get, it might be Questa version specific, I will switch the version and try to fix it.

> GUI mode works fine too, but you might want to consider supporting Visualizer. 

I added a visualizer support now. The `modelsim` function now has 2 arguments:
```
modelsim(${IP} GUI)   # launches legacy GUI
modelsim(${IP} GUI_VISUALIZER) # launches visualizer
```

For you to test it nothing should change.

> Compile needs some more options. I was thinking how to allow user to specify some custom compile/run options. Should that all go to CMakeList.txt?
[... 13 lines cut ...]

This would then let you specify these additional arguments with `cmake -DSV_COMPILE_ARGS="...."`.
But I would strongly recommend to modify the `CMakeLists.txt` file.
My argument is that you want to keep the simulation as stable as possible, letting users only to enable/disable GUI, change some simulation parameters etc...
[... cut ...]
** Error: (sccom-6207) No object files found for SystemC version 2.3.4 with gcc-10.3.0 in any library. Make sure that compilation and linking is being done with correct options.
```

I managed to reproduce it, and for me it was problem with `-32` and `-64` compilation.
I now added an explicit arguments to each step, by default it uses `-64`, if you want to use `-32` you can call modelsim with
```
modelsim(${IP} 32BIT)
```

Hopefully, this will solve the issues you saw.
Thanks for reporting it,  there are always corner cases on different OS-es and machines, hopefully, we manage to detect most of them.

Risto97, 2025-03-05:
P.S.

Another tip for you that can help you debug SoCMake and CMake in general.
When running any of the targets with make, you can pass `VERBOSE=1` like this:

```
make run_test_counters_modelsim VERBOSE=1
```

And this will print for you every command that make executes

Risto97, 2025-03-06:
I have added also the XCelium support.
It is on a branch https://github.com/HEP-SoC/SoCMake/tree/xcelium_sc_sv

The example became a bit convoluted, mostly because currently all 3 simulators work differently.
Although Xcelium and OSCI work similarly, and this is in general how I would like to have support done for all simulators.

1. XCelium SystemC libraries are compiled fully with GCC, and later loaded in simulation with `-loadsc` flag. Libraries are linked to SystemC distribution contained within xcelium simulation
2. The OSCI simulator has to contain `sc_main()`, it is compiled with GCC, and linked to the user-provided SystemC OSCI library. In case the DUT is in Verilog, Verilator is used to translate to SystemC
3. Questasim simulation cannot use GCC for now, but is using `sccom` to compile SystemC files, so the libraries are added with SoCMake's `add_ip()` instead of standard CMake libraries for C++ with `add_library()` like other two. Hopefully, I could make it work more similar to XCelium in the future.

All the test cases work on my machine, but I do expect that there might be issues on another OS and another machine, I will try to run it on another machine to sort out some corner cases.
Let me know if you find time to test it out.

I will start working on VCS next, and then hopefully find a way to consolidate all 4 simulators to work similarly.

Risto97, 2025-03-10:
Added support for VCS mixed language simulation on [vcs_sc_sv](https://github.com/HEP-SoC/SoCMake/tree/vcs_sc_sv) branch.

The flow compiles SystemC libraries with GCC only.
I am using `vlogan` and `syscan` to generate SystemC and Verilog wrappers.
This is done automatically only for libraries where the language crossover happens.
I managed to generate Verilog wrapper with `vlogan` from a shared object previously compiled with GCC, which means that everything can pretty much be compiled with GCC.

I will revisit Questasim, to hopefully get it to work with GCC as well.

amal-khailtash, 2025-03-10:
This is great @Risto97.  It will probably take me some time to test these.

Risto97, 2025-03-23:
Sure whenever you have time.

I am merging #129  into `develop` which provides mixed language SystemC and Verilog/SV simulation for 4 simulators:
Xcelium, Questa, VCS, Verilator (limited).
@benoitdenkinger you can proceed with the merge from `develop` to `flows` branch.

There is still a TODO list left:

- [ ] Verify donuts topologies work SC-SV-SC, SV-SC-SV (provide an example)
- [ ] SC-VHDL, VHDL-SC, need to be tested (provide an example)
- [ ] Allow picking a SystemC version, as currently for some simulators (VCS), the version of SystemC is fixed
- [ ] Allow user to pass port conversion tables for generation wrapper shells (Verilog -> SC), define which SC types to use (bool, sc_uint, sc_lv, ...). Possibly invent a simulator agnostic format for this purpose.
- [ ] Investigate Vivado simulator SystemC capabilities
- [ ] Test if signal tracing (debug) works for SystemC model compiled with GCC for all simulators in GUI mode
- [ ] Test UVM SystemC for commercial simulators